fawkesrobotics/fawkes #20: Full code base reformatting

timn, 2018-11-07:
While we did have a code style guide, there are many differing styles in the various libraries and components. It would be useful to unify this to a common codified format, which can ideally be automatically produced and enforced.

In the timn/reformat branch, we did introduce such a format based on clang-format. It seems to be the best tool at this point in time. Essentially, it codifies smart tabs and basic code patterns.

This issue is mostly to discuss how to proceed with the reformatting, there are three principal options:
1. do a full re-formatting of the entire code base and for each and every commit (as if the code base had always been in the envisioned shape). This approach has several drawbacks. It will invalidate all current hashes, especially causing problems for domain repos which reference these hashes. All working copies become essentially useless, if not harmful.
2. do a full re-formatting on the *current* code base and create commits with the format changes per component.
3. only re-format code which we actually touch from now on forward.

I think 1. is infeasible. 3. will leave us with a lot of code not in the new format. Therefore, 2. is currently my preference, but I'd like to get some feedback first.

morxa, 2018-11-12:
I agree that (1) is not an option.

One drawback of (2) is that it will be more difficult to get the history of specific lines of code, e.g., `git blame` will not work anymore without a hassle. I use this a lot to see why certain changes were done, so I think this is a major drawback.

On the other hand, (3) requires a bit more work, because we will need to check each diff whether it was formatted correctly. But this should be possible with [`clang-format-diff.py`](https://github.com/llvm-mirror/clang/blob/master/tools/clang-format/clang-format-diff.py). It also requires some more work from developers, as they need to (manually) reformat modified code without reformatting the whole file. It's not difficult to do, but it is possible to do this wrong.

timn, 2018-11-13:
Thanks, @morxa, for the feedback.

For (3), you need to modify each file you touch, otherwise you could have inconsistent formatting within a single file. Even then, you probably want to extend to the full library or component, or otherwise you end up with inconsistent formatting within that module.

Blaming may take one additional step, checking out a pre-formatting version. However, with the many other changes we are currently making to improve the code base (e.g., #15, #21) this can already hit you easily in many places.

I think it is better to take the disruptive step once and get it over with, so that we have a clean slate going forward. I'll file a part 2 once #21 is merged, then we can have a look.

closed 2019-03-19 by timn