fawkesrobotics/fawkes #37: Locally build gtest from external sources

morxa, 2018-12-06:
I'm trying to enable tests on both Fedora and Ubuntu builds. While Fedora builds googletest as a shared library, Ubuntu only ships the sources, because gtest is mainly intended as a copylib. I'd like to build those sources locally and use that library. I'm seeing three options:
1. Provide a special recipe for gtest, something like:
    ``` Makefile
    $(LIBDIR)/test/libgtest.so: $(GTEST_SRCDIR)/gtest-all.cc
    	$(CC) $(CFLAGS) -c $(GTEST_SRCDIR)/gtest-all.cc -o gtest-all.o
    	$(CC) $(LDFLAGS) -o $(LIBDIR)/test/libgtest.so gtest-all.o $(addprefix -l,$(LIBS_libgtest))
    
    test: $(LIBDIR)/test/libgtest.so
    ```
    The disadvantage: We're not making use of any of the generic rules (e.g., `%.o: %.cpp`), so it's not very integrated into the build system.
1. Build gtest in the same way as any other library. However, this currently does not work, because it would create the object files in `$(GTEST_SRCDIR)` which may be read-only (e.g., because it's in `/usr/src/googletest`). So this would need some adaption of the build system. This may be hard though, because the object files need to be in a different location than the src files, so any rule of the form `%o: %.cpp` will not work.
1. Copy the sources into the local directory and then build as a normal library. The disadvantage is of course that we copy all the source files into the local tree. The advantage of this option: If there is no source anywhere on the system, we could fetch the source from upstream and then build it in the same way.

So I'm not sure what the best option is, or if there is a better way to do this. What do you think?

morxa, 2018-12-06:
I also tried breaking (1) into several steps (create object files first, then link in a separate step). However, that does not work because the generic rules in `rules.mk` override my rules.

timn, 2018-12-07:
Would it make sense to provide a PPA that has gtest in a way similar to Fedora?

Otherwise we can have a look and discuss this at some point.

morxa, 2018-12-08:
I don't think so, the [package description](https://launchpad.net/ubuntu/bionic/+package/googletest) says:
> NOTE: This package does not contain a library to link against, but rather
 the source code to build the google test library; this is because the google
 test library must be built with the same flags as the C++ code under test.

timn, 2018-12-08:
What I meant is build a custom package that does build the library that we need.

timn, 2018-12-08:
I think a custom package could solve this rather easily and since we do this in a PPA it doesn't matter for the Ubuntu devs think about this. As long as it gets our tests running we're good. And since our primary platform handles it nicely that way, we should give this a try, rather than spending even more effort just to compile this ourselves within our system. Our build flags are rather conservative anyway these days, so I don't see an issue there.

morxa, 2018-12-09:
I don't really like that idea, it means maintaining another repository for only a small benefit, and it's not very clean either. I'd prefer a solution in the build system.

timn, 2018-12-10:
I'm for the PPA, low maintenance, no adaptations if they change the project layout, no clutter in our code base. And it is already proven to work nicely on Fedora.

timn, 2019-01-14:
I have taken the liberty to rebase your branch and add the necessary bits to compile googletest from the Ubuntu directory. One thing that is still missing which would be nice to add is to check that the headers are actually available (in `src/libs/googletest/Makefile`) and print a warning if they are not (and fail tests). It would be nice if you could have a look into that?

We also need to fix #43 before this can be merged.

timn, 2019-01-14:
I have additionally seen that the clips-executive tests are not executed by default. The reason is that they are in the clips sub-dir, which itself has nothing to build. The proper course of action would be to move the `tests` on level up to `clips-executive`. It also seems that the tests are currently broken.

timn, 2019-02-13:
This is work in progress in #53.

closed 2019-02-27 by morxa