fawkesrobotics/fawkes #43: Invalid mutex usage in syncpoints

timn, 2019-01-11:
The feature `SyncPoint::lock_until_next_wait()` (and related code) makes invalid use of mutex locking. This was revealed by running the unit test (after making it compatible) on FreeBSD. https://github.com/fawkesrobotics/fawkes/blob/331cfd9c5b3ab831ed0693421f1f88b37f6492e3/src/libs/syncpoint/syncpoint.cpp#L415-L426
The code locks a mutex, that is then later unlocked during wait. However, these two functions are almost certainly invoked by different threads. However, the [POSIX documentation](http://pubs.opengroup.org/onlinepubs/009695399/functions/pthread_mutex_lock.html) states:
> Attempting to unlock the mutex if it was not locked by the calling thread results in undefined behavior.

That means that with the current way we run into this undefined behavior (we had a similar flaw in 0ba02108d184ba8a42860785c3f0c163dce48e5d 10 years ago). On Linux this does behave nicely, because Linux is forgiving this error. However, on FreeBSD (and potentially other POSIX platforms, e.g., Mac OS X) does not.

The proper way to handle this is to have an indicator variable, e.g., a bool. Then, the lock is only used to guard this variable. Instead of stopping by the mutex, the emit call must then use a wait condition for the value to change to the "is free" value (it may just use the `emit_locker_` variable and check for `empty()`).

timn, 2019-01-14:
The fix looks good. However, the run-time is really awful now, probably due to the increased timeouts? It takes 37 sec to run the SyncPointManagerTest on FreeBSD, but less than 16 sec on Linux. This could be improved by not relying on `pthread_tryjoin_np()`, but rather using a shared indicator variable that can be queried for completion. If you used `fawkes::Thread` this would be as easy as querying whether the thread is `running()` (busy in `loop()`). Can you improve that, please? Otherwise tests become an even worse burden on build performance.

timn, 2019-01-28:
@morxa, could you have a look at this?

morxa, 2019-02-06:
Using `pthrad_tryjoin_np()` is not really the problem here (and using `fawkes::Thread` won't help), because the test doesn't use it to wait for the thread, but instead just to check its status. If we use the thread's `running()` instead, we'd still need to wait some time before checking, to make sure that it had enough time to start.

I can try to reduce the wait times though, some of them may be unnecessarily long.

commit 55bfa721 (2019-01-14): syncpoint: use wait condition for lock_until_next_wait

Instead of using a mutex in a non-POSIX way to coordinate waiter and
emitter, use a wait condition.

This fixes #43.
commit a763d4c8 (2019-02-27): syncpoint: use wait condition for lock_until_next_wait

Instead of using a mutex in a non-POSIX way to coordinate waiter and
emitter, use a wait condition.

This fixes #43.
closed 2019-02-27 by morxa