fawkesrobotics/fawkes #59: Skill GUI low contrast when using dark theme

Kofferkulli, 2019-02-21:
When using the `Adwaita-dark` theme under GNOME it is nearly impossible to read the log screen of the `skillgui`. Especially the warning messages with yellow background are not readable. 

Example of warning messages with white font on yellow background: 
[image]

Example of error messages with red background:
[image]

timn, 2019-02-27:
That indeed looks barely readable. I think we need to cover both cases. The log viewer performs coloring of log entries, so this will be the source of the problem. A question is whether we can determine these colors automatically from the theme? This could also simply be configurable or automatically adjusted. There are already config options (see the code) which could be amended to enable manually switching to a dark theme. Even better would be if this can be automatically detected.

I'm happy to review a PR with your suggestions.

vmatare, 2019-03-29:
I'd say that "stale" bot is a little overzealous. This is a long-standing but low-priority issue and I don't think the `wontfix` label is appropriate at all.

timn, 2019-03-29:
Either someone fixes it now, or nobody cares and we close it. No overlong running tickets.

closed 2019-05-05 by stale[bot]