fawkesrobotics/fawkes #73: buildsys: rosversion is extremely slow on Fedora 29

vmatare, 2019-03-19:
This is significantly slowing down builds with nothing (or little) to do, including `make clean`.

```
# F29
> time rosversion roscpp
1.12.14

real  0m0,366s
user  0m0,296s
sys 0m0,067s

> rpm -qf $(which rosversion)
python2-rospkg-1.1.7-1.fc29.noarch
```
```
# F27
> time rosversion roscpp
1.12.12

real  0m0,069s
user  0m0,054s
sys 0m0,014s

> rpm -qf $(which rosversion)
python2-rospkg-1.1.0-2.fc27.noarch
```

Might be related to the Python 3 switcheroo on Fedora 29?

timn, 2019-03-19:
As my investigation results you pasted indicate, these are still Python 2 packages and tools. The rosversion script explicitly uses `/usr/bin/python2` as shebang.

timn, 2019-03-19:
As I further stated:
Nothing in https://github.com/ros-infrastructure/rospkg/compare/1.1.0...1.1.7 particularly sticks out as a reason...

It might be worth checking call sites for rosversion and see if they can be reduced, or if this doesn't help consider caching the result.

timn, 2019-03-19:
@vmatare, what lead you to the conclusion that `rosversion` was the culprit? I have checked and there are only few call sites and even if replacing the `rosversion` invocation altogether the overall slowness remains the same. I think something else causes builds to be much slower (though I cannot tell what so far).

timn, 2019-03-19:
Some more data. An an already-clean fawkes on current master branch.
```
# F29, working dir is on SSD
> time make clean
--> Entering sub-directory src/libs/core/qa
   --> Cleaning up directory src/libs/core/qa/
--> Entering sub-directory src/libs/core/tests
   --> Cleaning up directory src/libs/core/tests/
--> Cleaning up directory src/libs/core/

real	0m1,514s
user	0m1,017s
sys	0m0,606s
```
```
#F27, working dir is on NFS
> time make clean
--> Entering sub-directory src/libs/core/qa
   --> Cleaning up directory src/libs/core/qa/
--> Entering sub-directory src/libs/core/tests
   --> Cleaning up directory src/libs/core/tests/
--> Cleaning up directory src/libs/core/

real	0m0,868s
user	0m0,367s
sys	0m0,382s
```
So there definitely is an F29-related regression, but from what I can tell so far `rosversion` ain't it.

vmatare, 2019-03-20:
I suspected `rosversion` because it was the only other thing that I could continually observe using CPU in `top` during a recursive build. Maybe I can gather some more precise information with `strace -t` or something...

vmatare, 2019-04-11:
Turns out this is caused by the `environment-modules` package, which casually increases `bash` startup time tenfold. With `environment-modules` installed:
```
$ time bash -c exit

real    0m0,036s
user    0m0,026s
sys     0m0,010s
```
After `dnf erase environment-modules`:
```
$ time bash -c exit

real    0m0,004s
user    0m0,002s
sys     0m0,003s
```
Since makefiles tend to do a lot of that, we found our culprit. This will be fixed in the ansible branch `vmatare/fedora29`.

timn, 2019-04-11:
Good catch :+1:. I can confirm that removing that package causes tremendous speedups. The only thing that seems to need this is sphinx, which we don't need. So I guess we can close this issue then?

vmatare, 2019-04-12:
Well, I think it should be reported as a Fedora bug, and maybe mentioned in an install guide. But other than that, feel free to close it.

timn, 2019-04-12:
I have filed a Fedora bug at https://bugzilla.redhat.com/show_bug.cgi?id=1699234. I have updated our install documentation at https://trac.fawkesrobotics.org/wiki/FawkesOnFedora. Feel free to add this to other platform/domain-specific places as needed.

closed 2019-04-12 by timn