fawkesrobotics/fawkes #77: Robot-memory: Aggregate function is broken with mongodb > 3.6

Sagre, 2019-04-01:
The use of the aggregate function of robot-memory returns a document with ok status 0.0 and an error message: `The 'cursor' option is required, except for aggregate with the explain argument
`
https://stackoverflow.com/questions/47472688/spring-data-mongodb-the-cursor-option-is-required suggests that this is caused by a change in mongo db in version 3.6, enforcing the use of a cursor. I am currently lost how to fix this, but will continue to look further into it

morxa, 2019-04-01:
Yes, this depends on #16. I'm not saying that I will be fixed just by switching to mongocxx v3, but the relevant code will be replaced in #16 anyway.

morxa, 2021-05-27:
Won't fix this for now, it currently does not seem to be used.

closed 2021-05-27 by morxa