fawkesrobotics/fawkes #79: realsense plugin: F200 doesn't work on recent Linux

vmatare, 2019-04-08:
On any recent kernel >= 4.19 (in particular Fedora 29), the RealSense F200 doesn't work any more:
```
15:20:15.491037 RealsenseThread: Switch enabled
15:20:15.728307 RealsenseThread: No. of cameras: 1
15:20:15.728434 RealsenseThread:

Detected Device:
Serial No: 019150262708
Firmware 2.60.0.0
Name Intel RealSense F200
USB Port ID 4-2-1

15:20:16.733597 RealsenseThread: Realsense Error: UVCIOC_CTRL_QUERY:UVC_SET_CUR error 2, No such file or directory
15:20:16.733707 RealsenseThread: Realsense Error: �#P8
Aborted (core dumped)
```

Upstream bug:
https://github.com/IntelRealSense/librealsense/issues/3706

vmatare, 2019-04-08:
Quoting myself from upstream bug:

> looks like this is resolved by just trying again, which in our case didn't work because we were referencing free'd memory after `rs_free_error()` ;-)

vmatare, 2019-04-11:
The segfault causing Fawkes to crash is fixed by pull request #82, but it seems that isn't enough to make the F200 actually work. The point cloud data look weirdly wrong, I'm not even sure what I'm seeing when I look at it in rviz...

vmatare, 2019-04-12:
Progress: I can reproduce the problem with the example programs shipped with `librealsense-1.12.*`. Now `librealsense-1.12.*` is configured to use `v4l2` as hardware backend on Unix, whereas *only on MacOS* it will use `libuvc` to access the device. Now if I switch it to `libuvc`, the examples start working and I get proper depth image out of the depth stream. Will put forked/fixed librealsense on my github, and our Fedora 29 librealsense packages should then be rebuilt from that. `librealsense` upstream PR also coming up.

morxa, 2019-04-12:
Upstream PR is probably pointless as librealsense1 has not been supported anymore by upstream for a long time.

vmatare, 2019-04-12:
Well, it seems they are still merging PRs sometimes: https://github.com/IntelRealSense/librealsense/tree/legacy
So I put mine here: https://github.com/IntelRealSense/librealsense/pull/3749
Let's see what happens. In the meantime, @morxa please update the F29 packages from https://github.com/vmatare/librealsense/tree/legacy.

morxa, 2019-04-12:
OK, never mind then, let's see if they accept the PR.

Please test the Fedora update:
https://bodhi.fedoraproject.org/updates/FEDORA-2019-0ed9b6d207

vmatare, 2019-04-14:
So the `libuvc` backend turns out to have its own set of bugs:
1. The infrared stream needs to be enable for the depth stream to work (we could live with that)
2. Switching the device off seems to block indefinitely

The second one here is the actual problem. I'm not sure we can or want to keep the thing running all the time. Might have to do some calculations how much it would hurt battery life.

So last night I tried again to get the `v4l2` backend running, but the `librealsense` (legacy) backend implementation is a horrible mess and I haven't been successful so far. I guess that's one of the reasons why they did a complete rewrite in `librealsense-2`.

New battle strategy:
1. Try out the SR300 and see if complete replacement of our F200s (migrating to `librealsense2`) is an option
2. Remove the bundled `libuvc` from `librealsense1` and see if I can get it to switch off the device cleanly.

vmatare, 2019-04-14:
After a helpful comment from @ev-mp (one of the librealsense maintainers), I've gotten the depth stream to work. Currently it breaks the IR stream, but for us that doesn't matter. So the latest https://github.com/vmatare/librealsense might finally work for us...

vmatare, 2019-04-14:
OK, I tested it and now it's finally working cleanly. The fix is not ready for upstream yet, since it breaks but for our purposes it's good enough:
[image]

So @morxa, sorry about the back-and-forth, if you would once more update the fedora package from https://github.com/vmatare/librealsense, please...
Note however that right now it's just a preliminary hack that only fixes the depth stream.

morxa, 2019-04-15:
Thanks for digging into this.

F29 update: https://bodhi.fedoraproject.org/updates/librealsense1-1.12.3-b7a6fb9.3.fc29

morxa, 2019-04-19:
This is fixed with a librealsense patch, no need for changes on our side.

morxa, 2019-05-08:
F29 update: https://bodhi.fedoraproject.org/updates/FEDORA-2019-610a54f3fc

morxa, 2019-05-22:
Update is now in Fedora stable.

closed 2019-05-22 by morxa