fawkesrobotics/fawkes #110: realsense: add support for new realsense 2.0 hardware

goch, 2019-05-08:
realsense1 SDK is discontinued. Newer versions of the realsense Cameras use realsense2
https://github.com/IntelRealSense/librealsense

morxa, 2019-05-08:
Yes and no. Newer cameras need realsense2, older cameras still use realsense1. librealsense1 is still maintained in Fedora, the split was introduced with Fedora 28 (see [this Change Proposal](https://fedoraproject.org/wiki/Changes/librealsense2)).

That being said, if we switch to the SR300, we'll also need to switch to librealsense2.

morxa, 2019-05-27:
I created a Fedora Module that always contains the latest librealsense release, even for older Fedora releases. You can install it with:

    $ sudo dnf --enablerepo=updates-testing-modular module install librealsense/dev

That might be useful because librealsense upstream development is fast and the version in F29 is quite old (2.15.0, released in August 2018) and cannot be updated due to incompatible changes.

morxa, 2019-06-26:
Implemented in #132.

closed 2019-06-26 by morxa