fawkesrobotics/fawkes #124: Message desctructor coredumps now and then

momoson, 2019-06-16:
With the current master, `fawkes` coredumps from time to time.

The bug behind this lies in the Copy constructor of the message base class, which copies a linked list without adjusting the pointers pointing to the next element. By this, effectively only the first element of the linked list is copied, while the other elements remain as memory leaks.

The above described code part is already multiple years old. However, as only subclasses of the message base class are used, and they so far reimplemented the copy constructor without using the base class' one, the bug was not observable.

This was changed in commit (55c19e41319)[https://github.com/fawkesrobotics/fawkes/commit/55c19e41319e0546daf7619cee04a778ad347cc9], which started to utilize the base class copy constructor and hence, made the bug observable.

As a quick fix, the above commit can just be locally reverted, however, a proper fix should be implemented.

vmatare, 2019-06-17:
I took a closer look and I don't think the way the linked list is copied is the source of the problem. The `next` pointers are in fact updated correctly, the code that does it is just not very C-plusplussy. Currently, I cannot reproduce the problem, and indeed valgrind prints no warnings, regardless of whether I'm copying a message with just one or with multiple fields.

momoson, 2019-06-17:
Ah right, I did not dereference the double pointer correctly. My mistake, thanks for the update.

momoson, 2019-06-17:
After failing to correctly interpret the list copy I had to look a little bit deeper.

Interestingly, there is a common property of both cases in which the coredump happened. @goch reported the coredump for the SwitchInterface, while I saw it with the new tensorflow plugin. The messages which were employed to interact with these interfaces were in both cases field less.

Recreation of the coredump is fairly easy, as @goch described:
- The interfaces must be generated with the `ffifacegen` of the current master branch.
- Now start fawkes and load a plugin which implements one of the above interfaces. (I used the tensorflow plugin, see branch `msonnet/tensorflow_plugin` as it does not rely on any additional hardware.)
- Also load the skiller
- Send field less messages to the interface (E.g. with the `test_tensorflow` skill in `msonnet/tensorflow_test_skill`, see command TRIGGER_RUN)
- After a few messages, sometimes up to 20 or so, coredump should happen

Utilizing `gdb` I set breakpoints on the Copy constructor only if the message type was equal to the `"TriggerRunMessage"`. Comparing `mesg->fieldinfo_list_` with `fieldinfo_list_` shows normally both the expected value `0x0` for the empty list. However, just during the copy before the coredump, both values differ: The original message points to `0x0`, while the copied message points to an arbitrary memory position. As it does not point to an real object of type `interface_fieldinfo_t`, trying to delete the `next` element in this struct obviously fails. 

Reason for this behavior is the missing initialization of `fieldinfo_list_` in the case that the list is empty.

I am a little bit wondered why this wasn't already reported by static analysis of the code.

vmatare, 2019-06-17:
Oh, a fieldless message, didn't think of that. Good catch @momoson

morxa, 2019-06-18:
Good analysis and nice catch!

PR #131 (2019-06-18): Properly initialize fieldinfo_list_
This should fix the message destructor coredumps, thus resolve #124.

For trying: Use the description in the issue comment and spam the plugin with at least 60 messages.

No coredumps were notable anymore after applying this patch.
closed 2019-06-18 by morxa