fawkesrobotics/fawkes #161: Reconsider enforced auto-formatting

vmatare, 2019-09-19:
I'm starting to doubt whether enforcing clang-format is such a great idea after all. The way I see it, it's helpful to ensure a certain minimal quality of code layout. However I sometimes like to produce a high-quality code layout, and that is something that cannot be determined syntactically. A high-quality code layout always has to consider the code's semantics and the ergonomics of human vision, which is both (as yet) impossible for an auto-formatter.

Yes, it's good to have a minimum quality formatting, but it's not good to nail everything down to that minimum.

I'm not sure how we can have both. The first thing that comes to mind is giving a push warning and having stricter mergeability criteria if some branch violates the default formatting. We could also have some kind of "I know better than clang-format" marker that people can put in their files...

Opinions?

lyndwyrm666, 2019-09-19:
Maybe have it as some sort of review system? Teamleaders have the ability to approve code that is not compliant with the auto formatting?

morxa, 2019-09-19:
Yes, auto-formatting is about having a "pretty" formatting, but even more importantly, it's about having a consistent formatting. Good code formatting is subjective, a formatting that I find pretty may be suboptimal to you. But that's exactly the reason why we need to enforce a certain style. If we don't, the alternative is this:
1. I push some code with my preferred formatting.
2. You want to fix a bug. You don't like the formatting, so you have several options:
    1. Suck it up and stick with the original formatting. However, you don't actually know the rules of that formatting, so you may add inconsistencies with your changes.
    2. Reformat just the part that you want to edit, so it looks nice for you. Now, the formatting is inconsistent.
    3. Reformat the whole file. Obviously not a good option, because you will have a patch with hundreds of changed lines, just for a simple fix.
3. It's my turn again, as I want to add a new feature. So depending on your previous choice, I may have:
    1. Code that is formatted almost how I like it, but with some weird formatting, because you don't actually know the rules that I followed originally.
    2. A weird patch in the middle of the file that is formatted differently. But I'll just reformat that part, because it looks really funny to me.
    3. A file that looks completely different. What the f*** happened to my code?

As you can see, this easily ends in disaster (and this has actually happened). We will have a lot of commits that hide the actual changes behind formatting changes, while still ending up with a code base that is not formatted consistently.

Your comments only prove that point, as you suggest that there is one superior formatting. But your superior formatting is not necessarily mine, and neither of us is willing to just agree to the other's most favorite formatting choice.

Take-away message: consistency in formatting is important, and even more important than code that looks "pretty".

That being said, you can [tell clang-format to ignore certain parts of your code](https://philwrightdev.wordpress.com/2014/10/01/making-clang-format-ignore-sections/).

vmatare, 2019-09-19:
>  [...] you suggest that there is one superior formatting [...]
Well, yes and no. Certainly not in an abstract sense because code quality is not a one-dimensional concept. But which dimensions are most important varies by context.

It came to my mind because I wrote a [piece of code](https://github.com/fawkesrobotics/fawkes/blob/vmatare/gologpp/src/plugins/gologpp/exog_manager.cpp) that was **primarily** intended to be read by someone else (you btw ;-). In that case, readability is at the top of the soft criteria, and consistency (external, i.e. with other code), is far down (internal consistency is of course still important, but will often conflict with readability, which is again something that needs to be weighed by context).

I generally do agree that consistency is important, but it's definitely most important **within** one component. I don't really care if component X is formatted the same as component Y. The only problem with that is that people might have difficulty adapting to different styles when they patch something. But adapting (2.i. in your list) is really absolutely the only option.

Maybe I can work with that clang-format exception, but I'd prefer something more fine-grained, i.e. overriding/disabling only certain formatter settings locally...

vmatare, 2019-09-19:
The clang-format tool is a little dumb in the way it handles its config. Overrides don't seem to be implemented, but there's a clean API in the back, so it should be easy to implement...

vmatare, 2019-10-16:
Well, I've now tried two things. We can in fact have local `.clang-format` files in any source folder, but they cannot just override, they have to be complete definitions. Initially that looked good to me, but somehow the local format seems to have been ignored on some other devs' systems. Might warrant some investgation, and we should think about whether we want to allow that practice if we *can* get it to work.

The other thing is the `// clang-format off` comment, which is straightforward and less error prone.

I think my favourite solution would be to make some changes to the global format (mostly the alignment settings which often don't come out well) and then allow __sparse__ use of `// clang-format off`. Any comments?

closed 2019-11-23 by stale[bot]