fawkesrobotics/fawkes #172: buildkite linter is broken

vmatare, 2019-10-16:
Fails with:
```
[DOC] Building documentation (api-quick.doxygen). This may take a while...
--> Warnings have been generated:
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
error: Problems running dvips. Check your installation!
gmake: *** [/workdir/etc/buildsys/root/docs.mk:35: api-quick.doxygen] Error 1
🚨 Error: The command exited with status 1
```
I suspect it's a problem related to ghostscript. See if the buildkite hosts have ghostscript-9.27 since that seems to fail with the formulas generated by doxygen. Downgrading to ghostscript-9.26 solved the problem for me.

Doxygen issue: [https://github.com/doxygen/doxygen/issues/7290](https://github.com/doxygen/doxygen/issues/7290)

morxa, 2019-10-16:
Yes, I noticed that this afternoon, but I haven't had the time to fix it.

morxa, 2019-10-17:
This should be fixed in fawkesrobotics/docker-robotics@9ba20678caa84339f2e478794b53c8c869e7170c.

Waiting for the container builds to confirm.