fawkesrobotics/fawkes #174: SAC Segmentation of PointCloud finds 'no' lines (in simulations)

MostafaGomaa, 2019-10-20:
Environment:
Fedora 30 
Ros Melodic
Gazebo Simulation
---------
Segmentation of PointCloud into lines (in line-func.h) does not find any lines!

The segmentation calculation is used by LaserLinesThread to find laser-lines from laser point-cloud, as well as, by NavGraphGenerationThread to find obstacle lines in the map from obstacle point-cloud.

The issue could be detected on behavior level as robots repeatedly failing mps-align as it can not match with laser-lines to tag-lines (since there is no laser-line date to match).

I am wondering at this point if this workes on any other configuration or in the real world (since the issue should not be related to simulation as far as I understand)

morxa, 2019-10-20:
We haven't migrated the robots to Fedora 30 yet, so it may occur there as well.

MostafaGomaa, 2019-10-22:
It is currently suspected that the issue has something to do with pcl1.9. 
Pending on testing with pcl 1.8

morxa, 2019-10-22:
Yes, I can confirm that the issue does not occur with pcl 1.8.

MostafaGomaa, 2019-10-24:
An issue describing the problem has been filed 
https://github.com/PointCloudLibrary/pcl/issues/3439#issue-511898601

vmatare, 2019-10-24:
Great, so we can just patch with [36c2bd6](https://github.com/PointCloudLibrary/pcl/pull/2767/commits/36c2bd6209f87dc7c6f56e2c0314e19f9cab95ec)

morxa, 2019-10-24:
Already in progress: https://koji.fedoraproject.org/koji/taskinfo?taskID=38524013

morxa, 2019-10-24:
The patch indeed fixes the issue, I'm creating PRs for the Fedora packages.

MostafaGomaa, 2019-10-24:
Cool, thanks !

morxa, 2019-10-28:
There is now also a COPR until this has been merged in Fedora:
https://copr.fedorainfracloud.org/coprs/thofmann/pcl/

MostafaGomaa, 2019-10-29:
This has been tested resolved!
Thanks, @morxa and @vmatare.
I mark the issue as closed!

closed 2019-10-29 by MostafaGomaa