fawkesrobotics/fawkes #195: Build failure of gui target in F31 due to GLib >= 2.62

nlimpert, 2020-01-07:
The `gui` target fails to compile on Fedora 31 due to deprecation warnings coming from [changes with version 2.62 of `glib`](https://developer.gnome.org/glib/unstable/glib-Date-and-Time-Functions.html#GTimeVal).

Example output for `gui_utils` (note that `gui_utils` is marked as a subfolder for the `gui` target of `libs`):

```
# make -C src/libs/gui_utils/
make: Entering directory '/home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils'
[C++] fawkes/src/libs/gui_utils/service_selector_cbe.cpp
In file included from /usr/include/gdkmm-2.4/gdkmm.h:34,
                 from /usr/include/gtkmm-2.4/gtkmm.h:89,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_model.h:30,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:26:
/usr/include/gdkmm-2.4/gdkmm/pixbufanimation.h:124:72: error: ‘GTimeVal’ is deprecated: Use 'GDateTime' instead [-Werror=deprecated-declarations]
  124 |   Glib::RefPtr<PixbufAnimationIter> get_iter(const GTimeVal* start_time);
      |                                                                        ^
In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glibmm-2.4/glibmm/refptr.h:21,
                 from /usr/include/glibmm-2.4/glibmm/main.h:21,
                 from /usr/include/glibmm-2.4/glibmm/dispatcher.h:21,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/connection_dispatcher.h:28,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:25:
/usr/include/glib-2.0/glib/gtypes.h:549:26: note: declared here
  549 | typedef struct _GTimeVal GTimeVal GLIB_DEPRECATED_TYPE_IN_2_62_FOR(GDateTime);
      |                          ^~~~~~~~
In file included from /usr/include/gtk-2.0/gtk/gtkobject.h:37,
                 from /usr/include/gtk-2.0/gtk/gtkwidget.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkcontainer.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkbin.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkwindow.h:36,
                 from /usr/include/gtk-2.0/gtk/gtkdialog.h:35,
                 from /usr/include/gtk-2.0/gtk/gtkaboutdialog.h:32,
                 from /usr/include/gtk-2.0/gtk/gtk.h:33,
                 from /usr/include/gtkmm-2.4/gtkmm/box.h:44,
                 from /usr/include/gtkmm-2.4/gtkmm.h:91,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_model.h:30,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:26:
/usr/include/gtk-2.0/gtk/gtktypeutils.h:236:64: error: ‘GTypeDebugFlags’ is deprecated [-Werror=deprecated-declarations]
  236 | void            gtk_type_init   (GTypeDebugFlags    debug_flags);
      |                                                                ^
In file included from /usr/include/glib-2.0/gobject/gobject.h:24,
                 from /usr/include/glib-2.0/gobject/gbinding.h:29,
                 from /usr/include/glib-2.0/glib-object.h:23,
                 from /usr/include/glibmm-2.4/glibmm/value.h:24,
                 from /usr/include/glibmm-2.4/glibmm/error.h:22,
                 from /usr/include/glibmm-2.4/glibmm/iochannel.h:25,
                 from /usr/include/glibmm-2.4/glibmm/main.h:24,
                 from /usr/include/glibmm-2.4/glibmm/dispatcher.h:21,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/connection_dispatcher.h:28,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:25:
/usr/include/glib-2.0/gobject/gtype.h:685:3: note: declared here
  685 | } GTypeDebugFlags GLIB_DEPRECATED_TYPE_IN_2_36;
      |   ^~~~~~~~~~~~~~~
In file included from /usr/include/gtk-2.0/gtk/gtktoolitem.h:31,
                 from /usr/include/gtk-2.0/gtk/gtktoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtkmenutoolbutton.h:30,
                 from /usr/include/gtk-2.0/gtk/gtk.h:126,
                 from /usr/include/gtkmm-2.4/gtkmm/box.h:44,
                 from /usr/include/gtkmm-2.4/gtkmm.h:91,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_model.h:30,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:26:
/usr/include/gtk-2.0/gtk/gtktooltips.h:73:12: error: ‘GTimeVal’ is deprecated: Use 'GDateTime' instead [-Werror=deprecated-declarations]
   73 |   GTimeVal last_popdown;
      |            ^~~~~~~~~~~~
In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glibmm-2.4/glibmm/refptr.h:21,
                 from /usr/include/glibmm-2.4/glibmm/main.h:21,
                 from /usr/include/glibmm-2.4/glibmm/dispatcher.h:21,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/connection_dispatcher.h:28,
                 from /home/nlimpert/dev/fawkes-robotino/fawkes/src/libs/gui_utils/service_selector_cbe.cpp:25:
/usr/include/glib-2.0/glib/gtypes.h:549:26: note: declared here
```
I'm not entirely sure whether this is an issue in Fawkes.

--------------
A workaround (partly taken from [here](https://tecnocode.co.uk/2019/08/24/gtimeval-deprecation-in-glib-2-61-2/)) is as follows:

1. Install `gtkmm30-devel` (note [this issue](https://gitlab.gnome.org/GNOME/gtkmm/issues/55) for some details).
2. Apply the following patch:
```diff
diff --git a/etc/buildsys/gui.mk b/etc/buildsys/gui.mk
index 386c27394..3db2ec8fe 100644
--- a/etc/buildsys/gui.mk
+++ b/etc/buildsys/gui.mk
@@ -72,7 +72,7 @@ ifeq ($(HAVE_HILDONMM)$(HAVE_HILDONFMMM),11)
 endif
 
 ifeq ($(HAVE_GTKMM),1)
-  CFLAGS_GTKMM     = -DHAVE_GTKMM $(shell $(PKGCONFIG) --cflags '$(PC_GTKMM)')
+  CFLAGS_GTKMM     = -DHAVE_GTKMM $(shell $(PKGCONFIG) --cflags '$(PC_GTKMM)') -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_60
   LDFLAGS_GTKMM    = $(shell $(PKGCONFIG) --libs '$(PC_GTKMM)')
   ifeq ($(CC),clang)
     CFLAGS_GTKMM += -Wno-overloaded-virtual $(if $(HAVE_GTKMM_2),-Wno-mismatched-tags)

```

However I think the patch only disables the deprecation warning.

morxa, 2020-02-04:
Have you looked into actually fixing the deprecation warnings by replacing `GTimeVal` and `GTypeDebugFlags`?

nlimpert, 2020-02-04:
Yes but I've considered the problem to be in packages depending on `glib` (`glibmm24-devel`,  `gtkmm24-devel`, `gconfmm26-devel` ).
That's why I considered to setup the build flag as a workaround until those packages would be patched.

Other than that I haven't found a way to patch this within Fawkes. Would you be able to show me a pointer where to look at?

morxa, 2020-02-04:
Oh, yes then disabling the warning is the way to go.

nlimpert, 2020-02-04:
I'll create a PR accordingly.

nlimpert, 2020-02-04:
I think we would have to replace the dependency of the `fawkes-devenv` package to `gtkmm24-devel` with `gtkmm30-devel` as well.
For the workaround I had to remove `gtkmm24-devel` with `rpm -e --nodeps gtkmm24-devel` - before installing `gtkmm30-devel`.

What do you think?
If you agree - are you able to change the `fawkes-devenv` package?

morxa, 2020-02-04:
What's the difference? Is `gtkmm30-devel` a newer version? I'm not really familiar with their versioning scheme.

But yes, I can update `fawkes-devenv`.

nlimpert, 2020-02-04:
Yes they only seem to differ in the version. In the end the according spec files ([`gtkmm24`](https://src.fedoraproject.org/rpms/gtkmm24/blob/master/f/gtkmm24.spec), [`gtkmm30`](https://src.fedoraproject.org/rpms/gtkmm30/blob/master/f/gtkmm30.spec)) simply point to different versions of `gtkmm` found [here](http://ftp.gnome.org/pub/GNOME/sources/gtkmm/).

I have checked some of the GUI tools (`skillgui`, `fflasergui`) linked against `libgtkmm-3.0.so` and `libgtk-3.so` and everything seems to work well - I can't see any issues yet.

morxa, 2020-05-06:
I updated `fawkes-devenv` to use `gtkmm30-devel`. The issue itself was fixed with #200.

closed 2020-05-06 by morxa