fawkesrobotics/fawkes #198: make 4.3 breaks `etc/buildsys/lua.mk`

vmatare, 2020-02-03:
The whole build process effectively stops dead at https://github.com/fawkesrobotics/fawkes/blob/ab07e932d6b516cb54a40c093ea2a1217e6d6613/etc/buildsys/lua.mk#L34 .
What happens is that the nested function expansion left of the '>' operator ends up empty, causing `cat` to wait for standard input. Not sure yet what the actual cause of the problem is since I'm having a hard time wrapping my brain around all this pattern replace magic going on there.

vmatare, 2020-02-03:
Anyways, easy workaround is downgrading to GNU make 4.2.*

morxa, 2020-02-11:
I can't find an obvious reason in the make 4.3 changelog:
https://lwn.net/Articles/810071/

I can't really reproduce it, even Fawkes Rawhide is still on make 4.2.

morxa, 2020-03-18:
Can you please test this again? It seems to be working now, here's a build with make 4.3:
https://koschei.fedoraproject.org/build/8082487

morxa, 2020-03-18:
another successful build:
https://koji.fedoraproject.org/koji/taskinfo?taskID=42606430

vmatare, 2020-03-18:
Huh. Problem still exists on my desktop PC. But I seem to remember that it mysteriously worked on my laptop, so it might be something system-specific. Though hard to imagine what. Will investigate further...

morxa, 2020-04-21:
@vmatare Closing, as this doesn't seem to be reproducible. Feel free to re-open if you can reproduce the issue.

closed 2020-04-21 by morxa