fawkesrobotics/fawkes #230: cx: wm-robmem-sync insufficient timestamp information is delete query

MostafaGomaa, 2020-05-27:
When a `delete` trigger is processed, the `clusterTime` of the triggered query is compared to `update-timestamp` slot of the `wm-robmem-syn-map-entry` fact to determine if this trigger is older than our timestamp or should we process it.

This problem is that clusterTime is not enough for this comparison since it only contains the information about the `sec`  of the trigger (without the ms). This causes us to ignore deletes if they happened later within the same `sec` by another member considering then as an older timestamp
`wm-robmem-sync-delete: received delete for /domain/fact/wp-on-shelf?wp=CCG1&m=C-CS1&spot=LEFT with older timetamp than our own `
`Additional-info: [wm-robmem-map-entry update-timestamp:(1590546500 362000) > trigger clusterTime:(1590546499 0) ] `

This issue becomes worst when  combined with #229

MostafaGomaa, 2020-05-27:
The problem that I am encountering is that I currently do not know a way to include a timestamp in the delete trigger.
the general formate for the trigger message is:
```
{
   _id : <ObjectId>,
   "operationType": <string>,
   "fullDocument": <document>,
   "ns": {
      "db" : <string>,
      "coll" : <string>
   },
   "documentKey": {
     "_id": <ObjectId>
   },
   "updateDescription": <document>,
   "clusterTime": <Timestamp>
}
```
In the case of the `insert`, the updated-timestamp is included in the fullDocument. As far as I understand, the `delete` trigger only has the documentKey but it does not contain the fullDocument. .
Is there a way to include the fullDocument in the `delete `trigger?

If the answer is no, Then can we we use a 'fake' insert with an empty|special fullDocument for 'deletion'?

@morxa what do you think?

morxa, 2020-05-28:
There is no fullDocument in a delete, because the document was deleted. That's why we use the cluster time instead.

One solution to this problem would be to not actually delete the document, but merely add a deleted flag to the document (and remove the value). This way, we can use the time stamp from the document.

I'm still wondering when you see this issue? If this happens, you usually have two agents modifying/retracting the same facts in parallel. How do you know which one you actually want to keep?

closed 2020-09-12 by stale[bot]