fawkesrobotics/fawkes #231: cx: wm-domain-sync, rapidly obsolete information about wm-fact-idx in wm-syn-map-fact

MostafaGomaa, 2020-05-27:
Context:
When a wm-fact is initially asserted, a wm-sync-map-fact is added to map the `(wm-fact-idx ?wfx)` to the corresponding `(domain-fact-idx ?dfx)`.

Issue:
The wm-fact-idx is not updated when the wm-fact is modified but the `id` remains the same.
This causes the mapping to carry obsolete information about the wm-fact-idx, when the wm-fact is modified by `wm-fact-key2id` or any other reason.

proposed solution:
modify the rule `wm-sync-worldmodel-fact-true` to fire even when `(wm-fact-idx ~0)`. @morxa would there be any side effects to that I cant see.

This issue did not cause any problems as we almost we only use the `wm-fact-idx` to when the `value TRUE|FALSE` slot is updated. Otherwise, we use the `wm-fact (id ?id` to determine if the `wm-fact` was removed (which remains the same after the problematic modification).

morxa, 2020-05-28:
I'm not sure I'm understanding the issue. Whenever the `wm-fact` changes, the `wm-fact-idx` is updated:
https://github.com/fawkesrobotics/fawkes/blob/80c9dedc365bed1abedf42e3eaf6b03c21dcd994/src/plugins/clips-executive/clips/wm-robmem-sync.clp#L266-L278

Could you give an example what exactly your problem is?

morxa, 2020-05-28:
After a discussion with @MostafaGomaa, this seems to be a bug here:
https://github.com/fawkesrobotics/fawkes/blob/80c9dedc365bed1abedf42e3eaf6b03c21dcd994/src/plugins/clips-executive/clips/wm-domain-sync.clp#L357-L376

closed 2020-09-12 by stale[bot]