fawkesrobotics/fawkes #237: Webview fails to build with libmicrohttpd 0.9.71

nlimpert, 2020-08-13:
In F31 and F32 `libs/webview` fails to build with `libmicrohttpd-0.9.71`:

```
[C++] fawkes/src/libs/webview/server.cpp
/home/robotino/fawkes-robotino/fawkes/src/libs/webview/server.cpp: In member function ‘void fawkes::WebServer::start()’:
/home/robotino/fawkes-robotino/fawkes/src/libs/webview/server.cpp:194:51: error: invalid conversion from ‘int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)’ {aka ‘int (*)(void*, MHD_Connection*, const char*, const char*, co
nst char*, const char*, long unsigned int*, void**)’} to ‘MHD_AccessHandlerCallback’ {aka ‘MHD_Result (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**)’} [-fpermissive]
  194 |                             WebRequestDispatcher::process_request_cb,
      |                             ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
      |                                                   |
      |                                                   int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**) {aka int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, long unsigned int*, void**
)}
In file included from /home/robotino/fawkes-robotino/fawkes/src/libs/webview/request_dispatcher.h:29,
                 from /home/robotino/fawkes-robotino/fawkes/src/libs/webview/server.cpp:29:
/usr/include/microhttpd.h:2428:45: note:   initializing argument 5 of ‘MHD_Daemon* MHD_start_daemon(unsigned int, uint16_t, MHD_AcceptPolicyCallback, void*, MHD_AccessHandlerCallback, void*, ...)’
 2428 |                   MHD_AccessHandlerCallback dh, void *dh_cls,
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[1]: *** [/home/robotino/fawkes-robotino/fawkes/etc/buildsys/rules.mk:234: server.o] Error 1
make: *** [/home/robotino/fawkes-robotino/fawkes/etc/buildsys/objsdir.mk:34: .objs_fawkes] Error 2
```
----------------------
A quick workaround is to downgrade to `libmicrohttpd-0.9.66` which builds fine.

MostafaGomaa, 2020-08-14:
Seems like they broke the C++ API in the new version
https://lists.gnu.org/archive/html/libmicrohttpd/2020-07/msg00011.html

It seems to me that for now we need to stick to the old versions until a path comes out

@morxa  could you maybe complie the libmicrohttpd-0.9.66,  as I am not really familiar with the Copr stuff

lyndwyrm666, 2020-10-04:
@TarikViehmann has a temp fix for it in https://github.com/fawkesrobotics/fawkes/tree/tviehmann/microhttpd-temp-fix

morxa, 2021-01-05:
This has been fixed in #245.

closed 2021-01-05 by morxa